christianh17/ioBroker.bydhvs #27: Adapter not displayed on 'Instanzen' Page

StephanTexxpro, 2021-05-07:
Installed the adapter by `npm install iobroker.bydhvs`

Unfortunatelly, the adapter is not displayed on the 'Instanzen' page (so no configuration is possible), although it shows up on the 'Adapter' page as expected.

I'm running admin v 5.0.15 in combination with Node v12.22.1 and jscontroller 3.3.8.

Any help is more than welcome!

christianh17, 2021-05-07:
Did you press the "+"-Button to create an instance?

StephanTexxpro, 2021-05-07:
That's it - thank you so much! BTW - if you need a wireshark capture of a HVM 19.3 (7 Blocks) communication feel free to contact me.

closed 2021-05-07 by StephanTexxpro

christianh17, 2021-05-07:
Of course i am interested an the wireshark capture. You find my mail-address in my github-profile.
Thanks!