christianh17/ioBroker.bydhvs #103: Components Liste - ggf. für Adapter Erweiterung nutzbar?

JBSullivan, 2022-02-13:
Hier eine neue Liste aus dem Photovoltaik Forum. Vielleicht ist ja etwas dabei, was zum "aufpimpen" des BYD Adapter taugt.
http://<IP-des-WR>/components/readable

christianh17, 2022-02-13:
Hi!
Leider klappt der Link nicht, auch eine Suche beim Photovoltaik Forum hat mich auch nicht weitergebracht. Kannst Du vielleicht den Link aktualisieren? 
Danke!

JBSullivan, 2022-02-13:
Sorry - siehe oben. Habe ich nicht mitbekommen das diese Zeichen >< etwas unsichtbar machen.

christianh17, 2022-02-14:
Darauf bin ich nicht gekommen, wäre ja auch zu einfach gewesen. Aber: Leider bekomme ich einen 401-error und vorher eine Abfrage nach Username und Kennwort. Was hast Du denn da eingegeben?

JBSullivan, 2022-02-14:
Nein, ich habe nichts eingeben müssen - nur die URL und dann kam schon die Mega lange Liste. Ich muss mich aber auch nicht auf meinem WR als Technican jedes mal neu anmelden. Ich musste das nur einmal machen, seit dem bin ich immer automatisch angemeldet. 

Hier mal ein Teil Screenshot von dieser Liste.

[image]

JBSullivan, 2022-02-14:
Ne, kommt über den WR, der aber in dieser Liste deutlich mehr Preis gibt, als was man über das Webinterface sehen kann - auch Batteriedaten, die der Gen24 ja gar nicht ausgibt.

Hier der Beitrag aus dem Photovoltaik Forum, für den Fall, das du es dort noch nicht gefunden hast.

https://www.photovoltaikforum.com/thread/165641-gen24-sw-1-17-2-1/?postID=2502035#post2502035