christianh17/ioBroker.bydhvs #105: Error Fehler

Trunks1982, 2022-02-24:
Hi in meinem Log kommt immer dieser Fehler siehe Bild.
Ich habe einen HVM Speicher mit dem der Adapter soweit auch funktioniert.

[image]
.

christianh17, 2022-02-25:
Das ist komisch, ich kann das mit meinen Tests leider nicht nachvollziehen. Kannst Du bitte mal den Log-Modus auf "Silly" setzen und zusätzlich in den Settings des Adapters den Testmodus aktivieren? Dann das Log herunterladen (nicht per Copy und Paste kopieren, dann fehlen Daten). Mit dem Log kann ich dann auf die Suche nach dem Fehler gehen.
Viele Grüße

Trunks1982, 2022-02-25:
[iobroker.2022-02-26.log](https://github.com/christianh17/ioBroker.bydhvs/files/8145313/iobroker.2022-02-26.log)

Hi ich hoffe das hilft dir weiter.
Gruß Christoph

christianh17, 2022-03-06:
Danke, habs gefunden. Ich habe dieses logging auf info heruntergestuft. Kommt in der nächsten Version. Es gibt gleich ein commit, dann wäre es zunächst nur in der Version bei github, ich hoffe es finden sich einige Tester dafür.

Trunks1982, 2022-03-06:
werde es testen

christianh17, 2022-03-06:
Danke!

Trunks1982, 2022-03-06:
Fehler scheint nicht mehr zukommen.Bis jetzt kein Eintrag mehr im Log.

closed 2023-02-20 by christianh17