christianh17/ioBroker.bydhvs #198: Mehrere Türme parallel

HarrysLapTimer, 2023-02-20:
Moin und Danke erst mal für die Arbeit!

Ich bin selbst kein iobroker Nutzer (bin auf openHAB) und würde den Code in einem ersten Schritt gerne nach Python migrieren und auf der Kommandozeile nutzen.

Frage: ich habe zwei 12.7er HVS Türme parallel. In BeConnectPlus kann ich BMS1 und BMS2 auswählen um die jeweiligen Zellinformation anzusehen. Hast Du eine Idee an welcher Stelle der request/response Kette ein zweiter Turm abgefragt würde? Oder bräuchte es dafür einen weiteren Wireshark-Mitschnitt?

christianh17, 2023-02-20:
Hi!
Da habe ich leider auch erst mal keine Idee. Ein Nutzer hat mir einen Mitschnitt vom Auslesen von zwei HVM-Türmen geschickt, da waren die ersten Pakete "hin und zurück" gleich. Danach gab es etwas andere "Querys". Ich hab das aber noch nicht fertig durchdrungen, fehlt mir gerade an Zeit dafür. Wenn ich soweit bin: Dann bekommst Du gerne eine Info.

Ich würde Dir empfehlen mit Wireshark ranzugehen und selbst einmal zu prüfen ob Du ein System erkennst. Wir könnten dann ja vielleicht auch die Querys tauschen - die Antworten die der andere Nutzer mitgeschnitten hat würde ich nicht weiterleiten, es sei denn er stimmt zu.

HarrysLapTimer, 2023-02-20:
O.k. Danke. Ich schaue jetzt erst mal ob ich an einen Turm ran komme. Habe aktuell noch irgendwie ein Netzwerk-Problem, beim ersten Request kommt immer eine leere Nachricht (0 Byte) zurück. Egal ob ich ein Fragment von Dir in node laufen lasse oder den ersten Request per Python socket schicke. Ich kann den Turm anpingen und bekomme auf dem Port 80 auch den Anmeldedialog (BYD HVS Netzwerk-Konfiguration), vom Port 8080 kommt aber wie gesagt noch nichts.

christianh17, 2023-02-20:
Machs Dir doch einfach und probier mal iobroker in einer VM aus, installiere meinen Adapter, IP-Adresse eintragen und schau in die Objekte. Wenn das funktioniert: Dann weißt Du, dass das Setting passt, wenn nicht musst Du an Netzwerk etc. basteln.

Tapter, 2023-02-20:
@christianh17: Du darfst meine snoops der 5-Modul HVS gerne weiter geben, wenn es hilft. Ich habe auch irgendwo noch meine Analyse dazu, wie das Proto bis zu 5 Modulen funktioniert, die ich bei Interesse gerne weitergeben kann (sie ist aber halt recht kryptisch :) ).
Im dern 9ten Response gibt es da bei mir am Ende viele nullen, gut möglich, dass hier die nächsten Module direkt anschliessen, oder auch möglich, dass man für den zweiten Turm ganz von vorne starten muss mit initial einem anderen Wert; kann das selber auch nicht ausprobieren da ich nur einen Turm hab.

christianh17, 2023-02-21:
@Tapter: Ich glaube nicht, dass der 5-Module-Mittschnitt hilft. Bei dem HVM-Beispiel was ich vorliegen habe gibt es gleich relativ früh andere Pakete.

forTT78, 2023-11-24:
Hallo @christianh17 , vielen Dank für den Adapter! :+1: Wäre es möglich dass auch 2 parallele Türme angezeigt werden könnten? Ich habe ein Setup mit 2x 10,2 HVS

christianh17, 2023-12-02:
Sorry, klar das geht. Aber ich habe leider im Augenblick keine Zeit den Adapter entsprechend zu erweitern :-(.

MK-2001, 2024-01-15:
@forTT78  könntest du bitte den Adapter prüfen, ich habe nur einen Aufbau mit 2x7,56. Da scheint es gut zu gehen. 
Meine Änderungen sind im PullRequest einsehbar. 

Die direkt URL, solange das es nicht in dieses Repo eingeflossen ist:
https://github.com/MK-2001/ioBroker.bydhvs/tree/master

closed 2024-01-16 by christianh17

forTT78, 2024-01-17:
@MK-2001 danke für die Info.

Zu Zeile 459 gibt es noch einen Anpassungsvorschlag: [Anzahl der Türme](https://www.photovoltaikforum.com/thread/215120-iobroker-code-und-unter-byd-auslesen/?postID=3563929#post3563929)

PR #262 (2024-01-16): Enable the analysis of two towers. [BREAKING]
introducing the structure 
bydhvs.0.Diagnosis.Tower_1. [...]
and
bydhvs.0.CellDetails.Tower_1. [...]

Closes #198 

If there is something missing, let it me know to merge this PR into 1.5.1