christianh17/ioBroker.bydhvs #204: hvsSOC (Diag) >undefined<

fboehri, 2023-07-02:
Bei der Abfrage der Werte eines HVM11.0 Systems ist der Eintrag hvsSOC (Diag) >undefined<, obwohl der hvsSOC Werte mit "71" belegt ist. Diese Info wird bei jeder Abfrage im Logfile als Info ausgegeben. 
Im Anhang das passende Logfile dazu. 

Viele Grüße
Frank

Logdatei
2023-07-02 21:07:15.365  - silly: bydhvs.0 (29165) hvsErrorStr     >no Error<
2023-07-02 21:07:15.366  - silly: bydhvs.0 (29165) hvsSOC (Diag)   >undefined<
2023-07-02 21:07:15.372  - info: bydhvs.0 (29165) undefined is not a valid state value for id "bydhvs.0.System.InvType"
2023-07-02 21:08:14.148  - silly: bydhvs.0 (29165) myNumberforDetails:62
2023-07-02 21:08:14.150  - silly: bydhvs.0 (29165) Poll start, IP:192.168.16.254
2023-07-02 21:08:14.263  - silly:  bydhvs.0 (29165) Received, State: 2 Data:  0103cc50303330543032305a32333034313032393238787878787803140000031b00001314010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000012345a02031b000c0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000000017070213080df804
2023-07-02 21:08:14.553  - silly:  bydhvs.0 (29165) Received, State: 3 Data:  0103320047014701470064001f51d6001d001a001b000003140000000000000c01000151ae00eb000000a8000000ff00004000042c6850
2023-07-02 21:08:14.832  - silly: bydhvs.0 (29165) Received, State: 4 Data: 0103061314010000001229
2023-07-02 21:08:14.834  - silly: bydhvs.0 (29165) NumCells: 64 Numtemps: 32 Modules: 4
2023-07-02 21:08:14.836  - silly: bydhvs.0 (29165) hvsSerial       >P030T020Z2304102928<
2023-07-02 21:08:14.837  - silly: bydhvs.0 (29165) hvsBMU          >V3.20-A<
2023-07-02 21:08:14.838  - silly: bydhvs.0 (29165) hvsBMUA         >V3.20<
2023-07-02 21:08:14.839  - silly: bydhvs.0 (29165) hvsBMUB         >V0.0<
2023-07-02 21:08:14.840  - silly: bydhvs.0 (29165) hvsBMS          >V3.27-A<
2023-07-02 21:08:14.841  - silly: bydhvs.0 (29165) hvsModules      >4<
2023-07-02 21:08:14.842  - silly: bydhvs.0 (29165) hvsGrid         >OffGrid<
2023-07-02 21:08:14.842  - silly: bydhvs.0 (29165) hvsSOC          >71<
2023-07-02 21:08:14.843  - silly: bydhvs.0 (29165) hvsMaxVolt      >3.27<
2023-07-02 21:08:14.844  - silly: bydhvs.0 (29165) hvsMinVolt      >3.27<
2023-07-02 21:08:14.846  - silly: bydhvs.0 (29165) hvsSOH          >100<
2023-07-02 21:08:14.846  - silly: bydhvs.0 (29165) hvsA            >3.1<
2023-07-02 21:08:14.847  - silly: bydhvs.0 (29165) hvsBattVolt     >209.5<
2023-07-02 21:08:14.848  - silly: bydhvs.0 (29165) hvsMaxTemp      >29<
2023-07-02 21:08:14.849  - silly: bydhvs.0 (29165) hvsMinTemp      >26<
2023-07-02 21:08:14.850  - silly: bydhvs.0 (29165) hvsDiffVolt     >0<
2023-07-02 21:08:14.851  - silly: bydhvs.0 (29165) hvsPower        >648.21<
2023-07-02 21:08:14.851  - silly: bydhvs.0 (29165) hvsParamT       >12.1<
2023-07-02 21:08:14.852  - silly: bydhvs.0 (29165) hvsBatTemp      >27<
2023-07-02 21:08:14.853  - silly: bydhvs.0 (29165) hvsOutVolt      >209.1<
2023-07-02 21:08:14.854  - silly: bydhvs.0 (29165) hvsError        >0<
2023-07-02 21:08:14.855  - silly: bydhvs.0 (29165) hvsErrorStr     >no Error<
2023-07-02 21:08:14.856  - silly: bydhvs.0 (29165) hvsSOC (Diag)   >undefined<
2023-07-02 21:08:14.858  - info: bydhvs.0 (29165) undefined is not a valid state value for id "bydhvs.0.System.InvType"

christianh17, 2023-07-03:
Der SOC auf den sich das bezieht ist der, der aus den Diagnose-Informationen kommt. Dieser hat auch eine Kommastelle und ist - auf dem "Papier" genauer aber wir wissen ja alle wie "genau" die SOC-Infos sind.
Insofern: Wenn die Diagnose-Informationen nicht eingeschaltet sind, wird der Wert nicht ermittelt und im log auf silly-Ebene gibt es eine entsprechende Anzeige.
Das möchte ich nicht anpassen, silly logging ist nur für die detaillierte Fehlersuche gedacht und nicht für den Regelbetrieb.
Viele Grüße!

fboehri, 2023-07-04:
Hallo Christian,
mir ging es um das Logging auf Info Ebene. Habe nur die silly Ebene in der Log Datei mit ausgeben lassen.

Dachte wenn der Eintrag "Diagnosis Data from Battery" deaktiviert ist, sollte nach meinem Empfinden auch kein Logging auf Info Ebene herauskommen, in dem auf undefined Daten verwiesen wird. Das Abfragen dieser Daten wurde ja per Konfiguration deaktiviert. 

Generell sehe ich natürlich die Entscheidung über das Logging bei dir.

Viele Grüße
Frank

christianh17, 2023-08-14:
Hallo Frank! Konntest Du das Problem nachstellen? Sonst würde ich den Issue schließen. Viele Grüße Christian

closed 2024-12-14 by christianh17