christianh17/ioBroker.bydhvs #205: Fehlende InvType IDs / Array

kago76, 2023-08-16:
Hallo Christian,

ich nutze eine HVS 12.8 mit BMS 3.27 und BMU 3.20 an einem SMA STP10.0 Smart Energy.
Beim setzen des States "InvType"  wurde ein " Error: undefined is not a valid state value" geworfen.

Grund:
Es gibt in der BeConnect 2.0.2 weitere Wechselrichtertypen - in meinem Fall "19"
Das Array hat aber nur die dir bekannten 17 Einträge ;-)

Diese möglichen WR Typen bekomme ich, wenn ich den Speicher der BeConnect mir anschaue:
```Fronius HV
Goodwe HV/Viessmann HV
Goodwe LV/Viessmann LV
KOSTAL HV
Selectronic LV
SMA SBS3.7/5.0/6.0 HV
SMA LV
Victron LV
SUNTECH LV
Sungrow HV
Kaco HV
Studer LV
SolarEdge LV
Ingeteam HV
Sungrow LV
Schneider LV
SMA SBS2.5 HV
Solis LV
Solis HV
SMA STP 5.0-10.0 SE HV
Deye LV
Phocos LV
GE HV
Deye HV
```

ID 19 passt für meinen daher.

christianh17, 2023-08-17:
Hallo Kago76!
Endlich habe ich meine Entwicklungsumgebung und mein Batterie-Simulationstool nach Rechnerwechsel wieder zum Laufen bekommen. Leider klappt es nicht so einfach mit der Erweiterung, offensichtlich wird der Invertertyp doch an einer anderen Stelle übermittelt als ich programmiert habe - jedenfalls zeigt beConnect etwas Anderes als mein Adapter an.

Daher: Kannst Du mir bitte einmal die Logausgaben (silly, und download, nicht copy und paste denn da fehlen dann Teile) einmal schicken? Damit bekomme ich das bestimmt kurzfristig hin.

Danke!
Viele Grüße
Christian

kago76, 2023-08-17:
Hallo Christian,

ich habe nur das Array myINVs
https://github.com/christianh17/ioBroker.bydhvs/blob/2cc32227499374c9f40fa77d8020838cc5ae8ca6/main.js#L141

um die o.g Werte erweitert, da mein BMS die ID 19 zurück bekommt.
Das Array hat aber keine 19 Einträge - daher der "Undefined" Fehler.

Grüße
Kai

kago76, 2023-08-18:
Hallo Christian,
die Änderungen an meinen eigenen System habe ich gerade hier "dokumentiert".
https://github.com/kago76/ioBroker.bydhvs/commit/597759bd8f052470aac0ad3e5e9a7bc10a1f343f

Da ich nur die Id 19 = "SMA STP 5.0-10.0 SE HV" bestätigen kann, kein pull request ;-)

closed 2024-02-22 by kago76