christianh17/ioBroker.bydhvs #207: Adapter stürzt nach Update auf js-controller 5 ab

florianru1984, 2023-09-18:
Nach Update auf js-controller 5.0.12 aus dem stable repository stürzt der bydhvs Adapter ab.

`<html>
<body>

bydhvs.0 | 2023-09-18 17:10:44.332 | warn | Terminated (UNCAUGHT_EXCEPTION): Without reason
-- | -- | -- | --
bydhvs.0 | 2023-09-18 17:10:44.332 | info | terminating
bydhvs.0 | 2023-09-18 17:10:44.329 | error | undefined is not a valid state value
bydhvs.0 | 2023-09-18 17:10:44.329 | error | Error: undefined is not a valid state value at Object.maybeCallbackWithError (/opt/iobroker/node_modules/@iobroker/js-controller-common/build/lib/common/maybeCallback.js:35:17) at AdapterClass._setState (/opt/iobroker/node_modules/@iobroker/js-controller-adapter/build/lib/adapter/adapter.js:5439:49) at AdapterClass.setState (/opt/iobroker/node_modules/@iobroker/js-controller-adapter/build/lib/adapter/adapter.js:5400:21) at setStates (/opt/iobroker/node_modules/iobroker.bydhvs/main.js:694:17) at Socket.<anonymous> (/opt/iobroker/node_modules/iobroker.bydhvs/main.js:827:17) at Socket.emit (node:events:514:28) at addChunk (node:internal/streams/readable:324:12) at readableAddChunk (node:internal/streams/readable:297:9) at Readable.push (node:internal/streams/readable:234:10) at TCP.onStreamRead (node:internal/stream_base_commons:190:23)
bydhvs.0 | 2023-09-18 17:10:44.328 | error | unhandled promise rejection: undefined is not a valid state value
bydhvs.0 | 2023-09-18 17:10:44.328 | error | Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch().
bydhvs.0 | 2023-09-18 17:10:38.606 | info | gestartet: 60 550
bydhvs.0 | 2023-09-18 17:10:38.606 | info | starte poll
bydhvs.0 | 2023-09-18 17:10:38.606 | info | BatDetailshowoften: 60
bydhvs.0 | 2023-09-18 17:10:38.606 | info | Bat Details : true

</body>
</html>`

christianh17, 2023-09-19:
Leider bin ich daran "gescheitert" auf meinem Entwicklungsrechner (Windows) ioBroker mit js-controller 5.0.12 zu installieren. Sobald ich ein Update durchführe läuft er nicht mehr - dubiose Fehlermeldungen, angeblich habe er ein Modul nicht gefunden (iobroker.js). Ich versuche es noch mal weiter. 
Bis dahin: Kannst Du bitte mal ein Debug log herunterladen (inkl. und mir schicken? Evtl. kann ich auch daraus erkennen welcher State mit undefined "beschrieben" werden soll und dagegen programmieren. Bitte im Adapter "Testmodus - Daten im Fehlerprotokoll anzeigen" anklicken und das Log herunterladen und nicht per Copy and paste herausziehen. Bei Copy and paste fehlen Teile.

Dee-16, 2023-09-22:
Hallo, 
ich erhalte noch Update auf keinen Zugriff mehr auf die BYD-Speicher. Der Wechselrichter, etc. hat Zugriff. Die IP stimmt auch.
Hängt das mit dem oben geschilderten Problem zusammen?

Ich verwende allerdings node: v18.18.0, js-controller: 5.0.12.

Meine Logs zeigen leider nicht viel:
2023-09-22 09:20:09.175 - info: bydhvs.0 (132195) starting. Version 1.4.0 in /opt/iobroker/node_modules/iobroker.bydhvs, node: v18.18.0, js-controller: 5.0.12
2023-09-22 09:20:09.235 - info: bydhvs.0 (132195) Poll Interval: 60
2023-09-22 09:20:09.237 - info: bydhvs.0 (132195) BYD IP Adress: 192.168.1.38
2023-09-22 09:20:09.239 - info: bydhvs.0 (132195) Bat Details : false
2023-09-22 09:20:09.239 - info: bydhvs.0 (132195) BatDetailshowoften: 60
2023-09-22 09:20:09.242 - info: bydhvs.0 (132195) starte poll
2023-09-22 09:20:09.243 - info: bydhvs.0 (132195) gestartet: 60 549
2023-09-22 09:20:09.754 - error: bydhvs.0 (132195) Error connecting to 192.168.1.38

unwichtig, 2023-09-23:
Hallo, 
hier das gleiche,  nach dem Update auf js 5.0.12.

[23-09-23-Log-bydhvs.txt](https://github.com/christianh17/ioBroker.bydhvs/files/12705773/23-09-23-Log-bydhvs.txt)

christianh17, 2023-09-23:
@unwichtig und @Dee-16 : Könnt Ihr bitte einmal testen ob es jetzt funktioniert? Ihr müsstet aber bitte den Adapter aus Github laden, sobald es bei Euch OK ist würde ich das Beta Repository updaten. Danke!

Dee-16, 2023-09-23:
@christianh17 
gerne als kleine Dokumentation.

Ich habe den Adapter deinstalliert und von GitHub entsprechend direkt installiert:

[image]

Dann sprechend die vordefinierte IP von auf meine 192.168.1.38 geändert.

[image]

**Und siehe da es läuft! - Besten Dank!!! :)**

[image]

Hier gerne noch für dich die Logs:

```
2023-09-23 19:03:27.516  - [32minfo[39m: bydhvs.0 (195318) terminating
2023-09-23 19:03:38.592  - [32minfo[39m: bydhvs.0 (195333) starting. Version 1.4.0 (non-npm: ChristianH17/ioBroker.bydhvs) in /opt/iobroker/node_modules/iobroker.bydhvs, node: v18.18.0, js-controller: 5.0.12
2023-09-23 19:03:38.660  - [32minfo[39m: bydhvs.0 (195333) Poll Interval: 60
2023-09-23 19:03:38.663  - [32minfo[39m: bydhvs.0 (195333) BYD IP Adress: 192.168.1.38
2023-09-23 19:03:38.664  - [32minfo[39m: bydhvs.0 (195333) Bat Details  : false
2023-09-23 19:03:38.665  - [32minfo[39m: bydhvs.0 (195333) BatDetailshowoften: 60
2023-09-23 19:03:38.668  - [32minfo[39m: bydhvs.0 (195333) starte poll
2023-09-23 19:03:38.669  - [32minfo[39m: bydhvs.0 (195333) gestartet: 60 549
2023-09-23 19:03:39.275  - [32minfo[39m: bydhvs.0 (195333) Received, Packet: 1 Data: 0103cc50303330543032305a32323032323231363938787878787803160314031b00011312020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000012345a01031b010b0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff00000000170917110413ddf5
2023-09-23 19:03:39.561  - [32minfo[39m: bydhvs.0 (195333) Received, Packet: 2 Data: 0103320061014a014a0064000e52a8001c0019001a00000316000000000000090200025276019100000120000000ff00004000042cec68
2023-09-23 19:03:39.840  - [32minfo[39m: bydhvs.0 (195333) Received, Packet: 3 Data: 0103061312020000009a6d

```

christianh17, 2023-09-24:
@Dee-16 Danke für den Test, ich weiß das zu schätzen :-).
Habe gerade das neue Release (1.4.1) in npm veröffentlicht, scheint aber zu dauern bis man es im ioBroker sieht.

florianru1984, 2023-09-29:
Bei mir funktioniert es mit der 1.4.1 auch

Dee-16, 2023-09-29:
@christianh17 ich kann leider über den iobroker nicht auf 1.4.1 oder 1.4.2 updaten. - Liegt es daran dass ich manuell auf 1.4.0 geupdatet hatte?!

florianru1984, 2023-09-29:
Bei mir ging's mit

`iob update beta
 iob upgrade bydhvs@1.4.2`

mcm1957, 2023-11-01:
> @christianh17 ich kann leider über den iobroker nicht auf 1.4.1 oder 1.4.2 updaten. - Liegt es daran dass ich manuell auf 1.4.0 geupdatet hatte?!
> 
> Edit: Hat sich fast erledigt, musste allerdings wieder manuell über den Link von Github installieren. Dann ging es....

Möcht nur bedingt ein mischen.
Aber du hast ziemlich sicher als Repository STABLE eingestellt. 1.4.2 ist noch nicht im stable.
Wenn du das Repository auf LASTEST = BETA eingestellt hättest, hättest du einen Update auf 1.4.2 ziemlich sicher bekommen.
Aber ACHTUNG: BETA ist Beta und daher für produktive System mit Vorsicht zu geniessen.

Eine Installation von Github sollte NUR in Abstimmung mit dem Dev passieren da auf Grithub jederzeit fehlerhafteste Zwischenversionen liegen können.

christianh17, 2024-01-16:
Thema erledigt :-).

closed 2024-01-16 by christianh17