christianh17/ioBroker.bydhvs #223: Update stable version in repo from 1.4.0 to 1.4.2

GermanBluefox, 2023-10-14:
# Think about update stable version to 1.4.2
**Version**: stable=**1.4.0** (347 days old) => latest=**1.4.2** (15 days old)
**Installs**: stable=**1068** (83.05%), latest=**99** (7.7%), total=**1286**

Click to use [developer portal](https://www.iobroker.dev/adapter/bydhvs/releases)
Click to [edit](https://github.com/ioBroker/ioBroker.repositories/edit/master/sources-dist-stable.json#L305)

Note: This is an automatically generated message and not personally authored by bluefox!

mcm1957, 2023-10-31:
@christianh17 

Is there any reason to delay release to stable any longer?

**If there is no comment until mid of November (15.11.) we might publish 1.4.2 at stable anyway.**

christianh17, 2023-11-01:
@mcm1957 To be honest: The requirements I have to do are difficult for me to achieve, so I did not take the time to dive in deeply. I am only "hobby"-programmer. If the only "leftover" is the splitting of the readme.md it should be no problem to do this and I could do a version 1.4.3 hopefully next weekend.

christianh17, 2023-11-01:
Thanks for the info about the readme.md and your offers to help me :-). Your offer cheers me up!

As far as I understand the comments from the users: the issue #207 is resolved with the new version. That issue was the reason for the new version ;-). And: Lots of people installed already the beta version, I think they did it because of the incompatibility between the new js controller and my adapter.

The dependapot gave me a green checkmark with the newest commit - I commited your PRs. 

What shall I do now? I really do not know what's the best to do.

You do not stress me - I know I should care about the new version going into the stable repository and sometimes I need a reminder.

Regards
Christian

mcm1957, 2023-11-01:
>And: Lots of people installed already the beta version
Not quite sure what you refer to. As you did not yet create a release which beta do you refer to? Direct github installation?

If you think your code is OK now the next steps would be to create a new release. Your installation of @alcalzone/releasescript seems to be incomplete. If it's ok I will create a PR to fix this.

christianh17, 2023-11-01:
It would be great if you could fix the releasescript with a PR :-). Thanks!

mcm1957, 2023-11-01:
I've created https://github.com/christianh17/ioBroker.bydhvs/pull/239

So the next steps would be
- merge the PR and check that no error are reported from tests
- update README.md and add a ne changelog section using tag ### __WORK IN PROGRESS__
- add your fixes to README.md
- add a note that now nodejs 16 is required as minimum to README.md
- create a new release by running 'npm run release'
  - please note that this hould be a minor release (1.5.0) minimum as the required node version is changed.
- check the results of the workflow.

If the woirkflow raises any errors and you do not see the reason at the first glance, feel free to contact me again.

mcm1957, 2023-11-01:
JUST HOLD ON - I just noticed that I missed update of pacakge-lock

christianh17, 2023-11-01:
Thanks again. Today I am too tired, I can't do anything reasonable any more.

mcm1957, 2023-11-01:
OK, PR is ready.

NO Problem - continue whenever you have time.
If I do not react please ping my again - I could likely miss some mail.

christianh17, 2023-11-04:
Hopefully all went good now, I managed with your help to create the new release. Thanks!
If there are some successful installs the next step will be a PR to get the release into the stable repo. :-)

mcm1957, 2023-11-04:
@christianh17 
Just noticed that you did not yet activate the automatic deploy to npm. (or it got lost during one of the latest updates)

I would suggest to uncomment the are at test-and-release.yml (https://github.com/christianh17/ioBroker.bydhvs/blob/c7e521208469de55ac94ce08c3393a0456c357bd/.github/workflows/test-and-release.yml#L79) unless you prefer manual deploment

1.5.0 is NOT yet deployed to npm and thus will not be available at latest. You may do manual npm deploy at the adapter directory or
- uncommnet the linked section
- ensure that a nom token is stores at github secrets (for instructions see referenced location at test-and.release)
- create a new 1.5.1 release ( or a 1.5.1-alpha.0 to test it before)

christianh17, 2023-11-05:
@mcm1957 : Thanks again, it is so kind from you to support me!
I did a manual deploy and for the future: I did as you recommended, so the next release will be pushed automatically to npm :-).
Today I wanted to install the 1.5 version on my productive ioBroker, but this has to wait till the new version will be accessible via the beta repository. Of course it worked on my test installation. 
Nice weekend!

mcm1957, 2023-11-05:
version should be available at latest repo  this afternoon

closed 2023-11-21 by GermanBluefox