christianh17/ioBroker.bydhvs #266: Add structure to CellDetails

MK-2001, 2024-01-16:
**Describe the bug**  
Currently all Temps / Volts are within one folder in ioBroker.Objects are listed down. For my behavior it is uncomfortable to review these values and determine which module is affected. 

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to 'Objects'
2. Click on 'bydhvs.x.CellDetails'
3. Scroll down to 'your value'

**Expected behavior**  
With the already determined information it should be possible to add an additional folder per module. 

**Versions:**  
 - Adapter version: 1.5.0

**Additional context**  
If you @christianh17  as adapter owner agree I will create a PR with this solution in it.

christianh17, 2024-01-16:
I am not sure, it breaks the scripts and logic if we change the structure of the objects. And: In my opinion the right way is not to use the objects "itself" but some kind of visualisation depending on the objects.
On the other hand: You did a great job in enhancing the adapter and I can see: Other people do use the objects itself.
What about a compromise: A switch in the configuration from the adapter and one can choose between the "old" (actual) behaviour and a more structured way for the cell details.

christianh17, 2024-12-21:
The issue is some days older, there have been changes. Do you think a change is still neded?