christianh17/ioBroker.bydhvs #268: Calculation of count towers

MK-2001, 2024-01-18:
@MK-2001 danke für die Info.

Zu Zeile 459 gibt es noch einen Anpassungsvorschlag: [Anzahl der Türme](https://www.photovoltaikforum.com/thread/215120-iobroker-code-und-unter-byd-auslesen/?postID=3563929#post3563929)

_Originally posted by @forTT78 in https://github.com/christianh17/ioBroker.bydhvs/issues/198#issuecomment-1896211922_

christianh17, 2024-02-04:
Hallo MK-2001!

Endlich hatte ich neben meinem Umbau zu Hause mal ein paar Minuten Zeit. Ich habe alles gemergt, aber leider bekomme ich immer noch fehler im Zugriff auf die towerAttributes. 

`TypeError: Cannot set properties of undefined (setting 'hvsMaxmVolt') at decodePacket5 (/opt/iobroker/node_modules/iobroker.bydhvs/main.js:647:46) at Socket.<anonymous> (/opt/iobroker/node_modules/iobroker.bydhvs/main.js:972:13) at Socket.emit (node:events:517:28) at addChunk (node:internal/streams/readable:368:12) at readableAddChunk (node:internal/streams/readable:341:9) at Readable.push (node:internal/streams/readable:278:10) at TCP.onStreamRead (node:internal/stream_base_commons:190:23)`

Ich schaffe es zur Zeit nicht, mich da intensiv einzudenken. Magst Du noch mal schauen? 

Offensichtlich bekomme ich keine Mails bei einem PR, kannst Du mir bitte zusätzlich noch eine Mail schreiben? Dann kümmere ich mich auch deutlich schneller als die letzten Male.

Danke für Deine Unterstützung!

Viele Grüße
Christian

MK-2001, 2024-02-05:
@christianh17 
Wie kann ich dir eine Mail schreiben? Ich denke so sollte es aber gehen. 

Zu deinem Problem: Wieviele Towers werden vom Adapter beim starten des Adapters gemeldet?
[image]

Eigentlich sollte beim Silly auch die Info kommen, ob er das leere Objekt erstellt hat:
[image]

Nur damit ich weiß, wann er in diese Zeile geht:
https://github.com/christianh17/ioBroker.bydhvs/blob/7eccca599bcf911bc3e1abf15fd4e34c87757024/main.js#L1140

Ggf. ist durch das neue Attribut im Admin dieses noch leer. Dann muss ich das abfangen...

christianh17, 2024-02-05:
Hi! 
Jetzt habe ich eine Info bekommen, funktioniert :-). Hier ein kompletter Zyklus des Adapter-Starts im Silly-Mode. 

Hilft das?
Viele Grüße
Christian

```
bydhvs.0
	2024-02-05 20:28:21.790	warn	Terminated (UNCAUGHT_EXCEPTION): Without reason
bydhvs.0
	2024-02-05 20:28:21.790	info	terminating
bydhvs.0
	2024-02-05 20:28:21.790	silly	got unload event
bydhvs.0
	2024-02-05 20:28:21.789	error	Cannot set properties of undefined (setting 'hvsMaxmVolt')
bydhvs.0
	2024-02-05 20:28:21.789	error	TypeError: Cannot set properties of undefined (setting 'hvsMaxmVolt') at decodePacket5 (/opt/iobroker/node_modules/iobroker.bydhvs/main.js:647:46) at Socket.<anonymous> (/opt/iobroker/node_modules/iobroker.bydhvs/main.js:972:13) at Socket.emit (node:events:517:28) at addChunk (node:internal/streams/readable:368:12) at readableAddChunk (node:internal/streams/readable:341:9) at Readable.push (node:internal/streams/readable:278:10) at TCP.onStreamRead (node:internal/stream_base_commons:190:23)
bydhvs.0
	2024-02-05 20:28:21.788	error	uncaught exception: Cannot set properties of undefined (setting 'hvsMaxmVolt')
bydhvs.0
	2024-02-05 20:28:21.788	silly	Received, State: 7, Data: 01038200800cd50cca416300130011011b00000000000000000000000000000000f25e0044fd780037000001f5106a00000118106700fe005e000000000000000018031903010050303330543032305a3230303331363731353578787878780209020900000cd30cd30cd30cd20cd30cd20cd30cd30cd30cd30cd30cd30cd30cd20cd20cd2145e
bydhvs.0
	2024-02-05 20:28:21.508	silly	Packet NOP
bydhvs.0
	2024-02-05 20:28:21.508	silly	Received, State: 6, Data: 01030288011f84
bydhvs.0
	2024-02-05 20:28:21.450	silly	Send, State: 6, Data: 010305510001d517
bydhvs.0
	2024-02-05 20:28:13.450	silly	waiting 8 seconds to measure cells
bydhvs.0
	2024-02-05 20:28:13.449	silly	Packet NOP
bydhvs.0
	2024-02-05 20:28:13.449	silly	Received, State: 5, Data: 0110055000024115
bydhvs.0
	2024-02-05 20:28:13.377	silly	Send, State: 5, Data: 0110055000020400018100f853
bydhvs.0
	2024-02-05 20:28:13.176	silly	NumCells: 128 Numtemps: 48 Modules: 4
bydhvs.0
	2024-02-05 20:28:13.108	silly	Received, State: 4, Data: 01030603140201010040ad
bydhvs.0
	2024-02-05 20:28:12.828	silly	Received, State: 3, Data: 010332001901480147005e0000a4240013001100110000031100000000000009020002a406b08000008f55000000ff00004000042c6be7
bydhvs.0
	2024-02-05 20:28:12.560	silly	Received, State: 2, Data: 0103cc50303330543032305a323030333136373135357878787878031003110319010103140201010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000180205162e172cf8
bydhvs.0
	2024-02-05 20:28:12.448	debug	connected set to true
bydhvs.0
	2024-02-05 20:28:12.446	silly	Poll start, IP:192.168.16.254
bydhvs.0
	2024-02-05 20:28:12.445	silly	myNumberforDetails:6
bydhvs.0
	2024-02-05 20:28:12.040	debug	connected set to false
bydhvs.0
	2024-02-05 20:28:11.948	silly	States system redis pmessage system.adapter.bydhvs.0.logLevel/system.adapter.bydhvs.0.logLevel:{"val":"silly","ack":true,"ts":1707161291945,"q":0,"from":"system.adapter.bydhvs.0","lc":1705522635408}
bydhvs.0
	2024-02-05 20:28:11.944	info	gestartet: 60 620
bydhvs.0
	2024-02-05 20:28:11.944	info	starte poll
bydhvs.0
	2024-02-05 20:28:11.944	silly	TestMode= true
bydhvs.0
	2024-02-05 20:28:11.944	info	BatDetailshowoften: 5
bydhvs.0
	2024-02-05 20:28:11.944	info	Tower count: 1
bydhvs.0
	2024-02-05 20:28:11.944	info	Bat Details : true
bydhvs.0
	2024-02-05 20:28:11.944	info	BYD IP Adress: 192.168.16.254
bydhvs.0
	2024-02-05 20:28:11.943	info	Poll Interval: 60
bydhvs.0
	2024-02-05 20:28:11.937	info	starting. Version 1.5.1 (non-npm: ChristianH17/ioBroker.bydhvs) in /opt/iobroker/node_modules/iobroker.bydhvs, node: v18.19.0, js-controller: 5.0.17
bydhvs.0
	2024-02-05 20:27:40.698	warn	Terminated (UNCAUGHT_EXCEPTION): Without reason
```

christianh17, 2024-02-05:
Jetzt gehts, danke!
Nun muss ich es noch irgendwie schaffen eine neue Version ins NPM zu bekommen, da hat es neulich Fehler gehagelt. Wird dann aber 1.6... 
Viele Grüße
Christian

MK-2001, 2024-02-05:
Hi @christianh17 
Ich kann dir gerne die Version 1.6.0 so vorbereiten, dass nur noch ein 
`npm pack`
Und
`npm push`
Notwendig ist. 

Aber ich habe nun selbst di aktuelle Version manuell installiert. Daher …

Gruß Marcus

christianh17, 2024-02-06:
Hi @MK-2001 (Markus) !
Heute hatte ich Zeit in die Objekte zu schauen. Mir ist aufgefallen, dass die zusätzlichen Informationen zum Tower nur im "Tower-1"-Ordner vorhanden sind, aber nicht in meinen Objekten. Damit meine ich Dinge wie Balancing oder ETA. Hat das einen speziellen Grund? Oder magst Du das vielleicht noch einbauen?
Das wäre Prima :-)
Danke und viele Grüße
Christian

MK-2001, 2024-02-06:
Hi @christianh17 ,
Ich dachte du hast nur einen Tower? 
dann würde ich aber nicht verstehen was du vermisst.
Gruß 
Marcus

christianh17, 2024-12-21:
Ich schaue mir das über Weihnachten an, war im Laufe des Jahres zu busy...