christianh17/ioBroker.bydhvs #287: The protocol is actually Modbus RTU tunneled over TCP

sarnau, 2024-04-03:
The protocol is simply Modbus RTU (the serial protocol) tunneled over TCP at port 8080.

To read data "holding register" are read (that is the $03 byte after the Unit-ID $01, which is the first byte). That's also the reason for the Modbus CRC at the end of the package. I would almost bet money on that the RS-422 Modbus port is using identical registers to communicate with the inverter. The registers value are 16-bit in MSB (high byte first), that is important e.g. for the serial number, which is stored in the registers 0 to 8 and therefore 18 ASCII characters long.

This also should make documenting it easier, because just the registers need to be documented – and not bytes in a buffer.

Here a minimal Python example to read the first 102 registers:

```
#!/usr/bin/env python3

from pymodbus.client.tcp import ModbusTcpClient as ModbusClient
from pymodbus.transaction import ModbusRtuFramer as ModbusFramer

client = ModbusClient(host='192.168.16.254', port=8080, framer=ModbusFramer)
client.connect()

print(client.read_holding_registers(0,102, slave=1).registers)

client.close()
```

sarnau, 2024-04-25:
I've documented my findings at https://github.com/sarnau/BYD-Battery-Box-Infos. I hope you'll find it helpful.

christianh17, 2024-12-21:
Thanks a lot, I will leave this issue open, so I can find this information easily in the future.

sarnau, 2024-12-21:
Glad it helps.