christianh17/ioBroker.bydhvs #293: Balancer Status

gallersd, 2024-07-06:
It would be good to also have datapoints for the balancer status of each cell in the diagnostics data.

The first data packet, which is read for the module data, should contain the bit-coded values for balancing the individual cells between the min/max voltages and the individual cell voltages. Usually 16 bytes with the value 0; when a cell is balanced, the respective bit jumps to 1.

BTW for my 8 modules HVM22 the adapter displays 128 cell voltages but only 64 temperature values ?!?

christianh17, 2024-07-16:
In my opinion it is a good idea to have the count of balanced cells. I just tried to implement it, I will see tomorrow if it works. If anybody else wants to try: Do an update from github, there is no new version number. The idea with the cell count came from krimhild (photovoltaik forum), thanks!
Regarding the temperatures: I think there a no more temperatures. Please do a check in beconnect. If there are more: let me now, please

gallersd, 2024-07-16:
I think the temperature observations from my side do not matter anymore.

The count of balanced cells is interesting and I will compare your values with my own programming ASAP.

My application of the count is to reduce the discharge power depending to the number of cells.
(e.g. dischargepower=1280/cellcount for a 128 cell tower)

christianh17, 2024-12-24:
Hi Gallersd, can I close the issue? Or are there still open problems?

gallersd, 2024-12-24:
You can close - for me all works perfect

christianh17, 2024-12-24:
Perfect, happy Christmas to you!

gallersd, 2024-12-24:
Thanks - same back to you …Gesendet mit der GMX Mail AppAm 24.12.24 um 16:46 schrieb christianh17
                
            
            
                
                    Von: "christianh17" ***@***.***>Datum: 24. Dezember 2024An: "christianh17/ioBroker.bydhvs" ***@***.***>Cc: "gallersd" ***@***.***>,"Author" ***@***.***>Betreff: Re: [christianh17/ioBroker.bydhvs] Balancer Status (Issue #293)
                
                
Perfect, happy Christmas to you!