christianh17/ioBroker.bydhvs #294: Balancer Status missing one Byte

gallersd, 2024-07-16:
BalancingOne consists of only 30 characters where it should be 32.

I assume line 
towerAttributes[towerNumber].balacing = data.slice(17,32).toString("hex");
must be changed to
towerAttributes[towerNumber].balacing = data.slice(17,33).toString("hex");

gallersd, 2024-07-16:
... of course same for tower 2:

towerAttributes[towerNumber].balacing_two = data.slice(17,32).toString("hex");

change to

towerAttributes[towerNumber].balacing_two = data.slice(17,33).toString("hex");

christianh17, 2024-07-16:
should be solved in github, can you please check?

gallersd, 2024-07-16:
I installed via IObroker from GitHub but version number did not change.
Shall I try in another way ?

Further problem is that I seldomly see  balancing - but will report ASAP ...

christianh17, 2024-07-17:
I did not change the version number, just changed the code. This should work as long as you install via URL.

gallersd, 2024-07-18:
My Balancer started today and for me the bug is fixed.
The additional BitCount value is great and my calculation results in the same value.
So I do not need my own calc anymore :-)

I only see a minor issue with the Object Name "tower state" to be identical for
BalancingCountOne
BalancingCountTwo
BalancingOne
BalancingTwo

gallersd, 2024-07-18:
BalancingBitCountOne
BalancingBitCountTwo
BalancingStateOne
BalancingStateTwo

zloe, 2024-08-01:
Wouldn't "BalancingCellCountXXX" be a more appropriate name?
It is the number of Cells that are being balanced, right?

christianh17, 2024-12-21:
siehe andere Kommentare

closed 2025-01-05 by christianh17