christianh17/ioBroker.bydhvs #309: Wrong name for Diagnosis.BatteryVolt and other mistakes in the object structure "Diagnosis"

zloe, 2024-08-01:
The "name" for the state Diagnosis.BatteryVolt is "Wirkungsgrad of that tower"

**To Reproduce**  
Open object browser and view the objects

**Expected behavior**  
The name should read something like "Voltage of Battery"

**Screenshots & Logfiles**  
[image]

**Versions:**  
 - Adapter version: Version from github (1.5.1?)
 - JS-Controller version: 6.0.9
 - Node version: 19.9.0
 - Operating system: Linux/Raspian

christianh17, 2024-12-21:
Kümmere mich drum, siehe auch Kommentar im PR

christianh17, 2024-12-24:
repaired in github, will be published in new version, thx for the hint!

closed 2024-12-24 by christianh17