christianh17/ioBroker.bydhvs #348: Intervall per Script setzen

Ralph1234567, 2024-12-15:
Hi,

wäre es möglich, das Intervall der Datenabfrage von "außen" änderbar zu machen (z B. ein Datenpunkt unter system.adapter.bydhvs, der per eigenem Script beschrieben werden kann)?

Hintergrund:
Ich logge die CellVoltages. An der Lade-/Entladegrenze ist dies nur sinnvoll möglich, wenn das Intervall klein genug ist, um die Änderungen zu erfassen. Das erzeugt allerdings auch dann eine entsprechende Datenmenge, wenn sich der Akku nicht an der Lade-/Entladegrenze befindet.
Idee deshalb: Bei entsprechend hoher bzw. niedriger Spannung Intervall kurz setzen, dazwischen Intervall lang setzen.

Viele Grüße
Ralph

christianh17, 2024-12-21:
Das ist eine coole Idee. Ich schaue mir das an - würde aber erst mal den PR bearbeiten. War ein wenig Pause mit dem Adapter da ich mit die Checks von Github nicht repariert bekommen habe, hatte aber gestern tolle Hilfe.

Ralph1234567, 2024-12-25:
Hi, frohe Festtage :-),

für mich würde das Einstellen der Abfragehäufigeit "an sich" sehr gut passen (also ein Datenpunkt, mit dem dynamisch das Intervall gesetzt werden kann, welches bisher fest in der Instanz gesetzt wird).

christianh17, 2025-01-05:
@Ralph1234567 Gerade habe ich eine Version hochgeladen in der Dein Wunsch erfüllt ist. - spätes Weihnachtsgeschenk. In der Konfiguration kann man jetzt anwählen, dass ein State verwendet werden soll, der State steht unter system (overridepoll). Kannst Du es bitte ausprobieren und ein Feedback geben? Danke!

Ralph1234567, 2025-01-05:
@christianh17 
Cool, danke schön  :-)
Hab ich gerade ausprobiert, funktioniert super.
Nochmals vielen Dank für die prompte Umsetzung 👍

closed 2025-07-24 by arteck